terjeio/grblHAL #211: Question about PPI

TadyTheFish, 2021-01-24:
Hello
I am trying to build a pin marking machine and I was browsing trough you code and I saw that there is code in the olugins/laser that should do the trick. Pulsing the output evenly when moving. But it seem that it doesn't work on the spindle pwm output.
Can you explain a little about that mode?

terjeio, 2021-01-25:
The plugin is just a frontend for a lowlevel driver implentation. Currently these implementations switches the spindle enable signal on/off. It should be fairly easy to add a compile time option to switch the PWM signal instead. If doing so then one should increase PWM frequency and check the response time of the PWM input of the machine to be controlled to see if is a viable option?

Note that only a few drivers supports this option. More can be added but there may be some that does not have a timer available to handle PPI mode.

> Can you explain a little about that mode?

Is [this explanation](https://www.engraversnetwork.com/support/universal-lasers/laser-how-tos/dpi-vs-ppi-laser/) ok?

TadyTheFish, 2021-01-25:
Where can I check what timer is used for this? I'm using STM32f411ce (blackpill).
$32=1, M112 P1
I have checked all spindle pins and nothing
I tried to send M3 and then move the machine. Still nothing

terjeio, 2021-01-25:
Is this with the latest test branch? There is a regression related to setting laser mode in that.

PPI requires laser mode enabled (`$32=1`) for PPI mode to become active.

TadyTheFish, 2021-01-25:
I managed to get it to work.. Thank you.. The problem was I didn't look at the spindle enable pin.
Could you please tell me how is PPI implemented? Is it a calculation from X and Y positions? Could you give me some code snippets?

terjeio, 2021-01-25:
It works by attching itself to the HAL step pulse output entry point and adding up travelled distance per step to the (next) PPI distance before pulsing the spindle enable signal. Pulsing of the enable signal is handled in _driver.c_.

https://github.com/terjeio/grblHAL/blob/2d871b2d0a71193df8015f98da1bc565d12341d0/plugins/laser/ppi.c#L65-L83

closed 2021-01-25 by TadyTheFish