terjeio/grblHAL #214: How to run the code on ESP32

krupis, 2021-01-27:
Hello, I have downloaded and setup my ESP-idf for this project, I have sucessfully flashes my ESP32 device with hello world program. I have downloaded this program folder but I cannot understand how do I compile it on my esp-idf. I cannot find any instructions either. 

I want to build  CNC controler and going to use ESP32 + Raspberry PI. Can someone give me some tips how do I compile and run the code on my ESP32 device.

terjeio, 2021-01-27:
cd to the _grblHAL/drivers/ESP32_ folder before using _idf.py_

The test branch has all the needed code in place, the master branch still needs at least the root grbl folder to be copied to the driver as the [Preparation step](https://github.com/terjeio/grblHAL/wiki/Compiling-GrblHAL#preparation) still applies.

krupis, 2021-01-27:
Thank you very much for a quick response. My apologies I did not mention that I am trying to run it using Eclipse. I followed the installation process and I think my Eclipse is setup properly. Are you familiar with Eclipse IDE?

I have opened the project as following:

[image]

At this point, I am not certain what is next

terjeio, 2021-01-27:
I am using Eclipse for all my projects, but just as an editor for the ESP32 and Arduino based drivers.

My attempts to compile/upload the ESP32 driver from Eclipse has failed, so I still have use _idf.py_. It would be nice to be able to do that from Eclipse instead...

krupis, 2021-01-27:
Thanks for answering. From the preparation steps as you have posted above, I have even tried to compile it on Arduino IDE but with no luck.

I have followed the step:
Load the selected grblHAL_xxx folder into the Arduino IDE as a Library.
Select the Sketch > Include Library > Add .ZIP Library menu. Add .ZIP Library supports both a .ZIP file or a folder. In our case, there is no .ZIP file.

[image]

I have went to project folder -> Drivers and selected ESP32, I have sucessfully added it as a library and now it appears on my arduino libraries folders

[image]

However, it does not show up under File -> Examples menu, perhaps maybe because the ESP32 is already one of the built-in libraries of the ESP32 devices?
[image]

Hoping to get some clarification on this.. 

In the meantime, I shall try terjeio method of compiling this code

krupis, 2021-01-27:
I am still not fully understanding how to compile it. What i have done so far:

1. I have copied the contents of root grbl folder to the drivers -> ESP32 -> grbl folder

I havent copied any of the plugins yet as I am not sure what I need yet. I am just trying to get it to compile. Can you clarify what you mean idf.py? I do not see such file in my ESP32 folder

terjeio, 2021-01-27:
You have to open the ESP-IDF Command Prompt (there should be a menu item for that in the start menu), cd to the driver folder and issue the _idf.py_ command. I compile and flash with _idf.py -p com74 flash_ where _com74_ is the com port of the ESP32 device.

See the [driver page](https://github.com/terjeio/grblHAL/blob/master/drivers/ESP32/README.md) for info and link to ESP installation details for WIndows.

krupis, 2021-01-27:
I dont think I have an ESP-IDF Command prompt. Since I am using macos, things are different here. I have followed this installation guide:

https://docs.espressif.com/projects/esp-idf/en/latest/esp32/get-started/

I think I failed at this step
[image]
Since I cannot figure out what is supposed to be  HOME

krupis, 2021-01-27:
Okay. Apparently, there is no need to replace HOME with anything, just cope and paste the command as it is. Now the terminal will allow me to execute idf.py build command. 
After executing the export.sh command, i CD into my ESP32 directory and run command :
idf.py build

[image]

I was able to build it, however, i received an error which I will try to solve now. Anyway, I am getting some results. Thank you very much!

terjeio, 2021-01-27:
> Perhaps you are fammiliar with the error:

Can't remember I have seen that error. Which version of esp-idf have you installed? If version 4 you may have problems as I understand it is not backwards compatible with version 3.3 that is used for this project.

On my machine there is a shortcut as I described above in my startup menu. It can also be found as a submenu to _ESP-IDF_ in the _All Programs_ list. I am using Windows 7.

krupis, 2021-01-28:
You might be right. I have found other people claiming having problems with version 4+ and I have installed the latest esp-idf version available. I will look into how to downgrade. 

From what I understood, for macos users, in order to ESP-IDF command prompt , you have to open a terminal and call the following command:
. $HOME/esp/esp-idf/export.sh

That will activate esp-idf command prompt for that terminal.

krupis, 2021-01-28:
I have updated my python to the latest 3.9 and downgraded the esp-idf version. I have deleted my previous esp folder and started over again with the installation of v3.3 esp-idf.

I have followed the steps:

https://docs.espressif.com/projects/esp-idf/en/v3.3/get-started/macos-setup.html

and 

https://docs.espressif.com/projects/esp-idf/en/v3.3/get-started/index.html#get-started-get-esp-idf

Step by step commands I have executed:

1. Install pip
'''
sudo easy_install pip
'''

2. Download toolchain for macos:
'''
 mkdir -p ~/esp
cd ~/esp
tar -xzf ~/Downloads/xtensa-esp32-elf-osx-1.22.0-80-g6c4433a-5.2.0.tar.gz
'''

First mistake I notice here :
tar -xzf ~/Downloads/xtensa-esp32-elf-osx-1.22.0-80-g6c4433a-5.2.0.tar**.gz**
.gz part is not needed since the file I have downloaded does not have .tar.gz  filename extension.

Calling command:
'''
tar -xzf ~/Downloads/xtensa-esp32-elf-osx-1.22.0-80-g6c4433a-5.2.0.tar
'''
sucessfully unzips the .tar file and puts it in my esp folder

3. Update path with command:
'''
export PATH=$HOME/esp/xtensa-esp32-elf/bin:%PATH
'''

I have completed the steps to setup the toolchain, now I move on to download and setup esp-idf v3.3

1. Clone 3.3v esp idf to esp directory
'''
cd ~/esp
git clone -b v3.3 --recursive https://github.com/espressif/esp-idf.git
'''

2. As suggested by espressif, I then need to Add IDF_PATH  as explained here:

Setup Path to ESP-IDF

The toolchain programs access ESP-IDF using IDF_PATH environment variable. This variable should be set up on your PC, otherwise projects will not build. Setting may be done manually, each time PC is restarted. Another option is to set up it permanently by defining IDF_PATH in user profile. To do so, follow instructions specific to Windows , Linux and MacOS in section Add IDF_PATH to User Profile.
https://docs.espressif.com/projects/esp-idf/en/v3.3/get-started/add-idf_path-to-profile.html#add-idf-path-to-profile-linux-macos

Using the command:
'''
export IDF_PATH=~/esp/esp-idf
'''

I assume I have to run this command manually everytime I want to use esp-idf?
However, they mention something like:
Set up IDF_PATH by adding the following line to ~/.profile file:
Unfortunately I was not able to locate .profile file anywhere.

3. I download the required python libraries :
'''
python3.9 -m pip install --user -r $IDF_PATH/requirements.txt
'''

Moving on to Start project part:
[... 95 lines cut ...]

After a quick google search I have figured out that I have to set an additional path for xtensa-esp32-elf
 with the following command:

'''
export PATH=$PATH:$HOME/esp/xtensa-esp32-elf/bin
'''

Now I am able to execute command
'''
make flash
'''
 for hello world example.

However, I am getting loads of security issues by my macos it does not allow me to execute this command due to
xtensa-esp32-elf-gcc: internal compiler error: Killed: 9 (program cc1)

There are many programs that are being stopped by macos security system. The list of programs:

cc1
as
collect2
Id

and others.. Is this normal behaviour? Can someone comment on that?

krupis, 2021-01-28:
I am getting another error now. I have setup everything as explained above, and was able to execute make commands just some security issues. I have now restarted my laptop and didint install any new software or change the configuration, and not able to run compile anything with make command anymore. If i understood corectly, everytime I launch terminal and want to compile esp-idf project, I must manually add paths as following:

export IDF_PATH=~/esp/esp-idf
and
export PATH=$PATH:$HOME/esp/xtensa-esp32-elf/bin

after that, I can cd to the project directory and execute make command. However, now I am getting a strange issue about make

[image]

Which I never seen  before

terjeio, 2021-01-28:
I have never used `make` for building ESP32 projects. What happens if you run `idf.py build` instead?

krupis, 2021-01-29:
idf.py is also not recognized but I think I know the issue. I have troubles with setting PATH and IDF_PATH

From what I understand, everytime I close/open console, i need to execute the following commands to set the correct paths as suggested by step 3:

Step 3. Set Environment Variables

ESP-IDF requires two environment variables to be set for normal operation:
https://docs.espressif.com/projects/esp-idf/en/release-v3.3/get-started-cmake/index.html#get-started-setup-path-cmake:
IDF_PATH should be set to the path to the ESP-IDF root directory.
**export IDF_PATH=~/esp/esp-idf**
PATH should include the path to the tools directory inside the same IDF_PATH directory.
**export PATH="$IDF_PATH/tools:$PATH"**

However, I remember seeing some other notes regarding PATH as described here:
https://docs.espressif.com/projects/esp-idf/en/v3.3/get-started/macos-setup.html

As you can see here, the PATH is not IDF_PATH/tools anymore, but
export PATH=$HOME/esp/xtensa-esp32-elf/bin:$PATH instead

I have not realised that there are 2 guides for installing esp-idf v3.3. I assume the only difference is one is to build using idf.py and the other using using make commands?

krupis, 2021-01-29:
Okay, so my latest findings attempting to build hello world using idf.py build. When I launch the terminal , I must add all 3 paths!:
```
export IDF_PATH=~/esp/esp-idf
export PATH="$IDF_PATH/tools:$PATH"
export PATH=$HOME/esp/xtensa-esp32-elf/bin:$PATH
```

After the paths have been added, I can cd into esp/hello_world and execute idf.py build command . However, I am getting some security errors as I have mentioned before:

[image]

krupis, 2021-01-29:
After allowing 10 programs due to security issues, i was able to proceed with idf.py build however another error occured:

```
Lukass-Air:hello_world Lukas$ idf.py build
Note: You are using Python 3.9.1. Python 3 support is new, please report any problems you encounter. Search for 'Setting the Python Interpreter' in the ESP-IDF docs if you want to use Python 2.7.
Checking Python dependencies...
Python requirements from /Users/Lukas/esp/esp-idf/requirements.txt are satisfied.
Running cmake in directory /Users/Lukas/esp/hello_world/build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 --warn-uninitialized /Users/Lukas/esp/hello_world"...
Warn about uninitialized values.
-- Found Git: /usr/local/bin/git (found version "2.30.0") 
-- IDF_TARGET not set, using default target: esp32
-- Building for target esp32
-- The C compiler identification is GNU 5.2.0
-- The CXX compiler identification is GNU 5.2.0
-- The ASM compiler identification is GNU
-- Found assembler: /Users/Lukas/esp/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /Users/Lukas/esp/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /Users/Lukas/esp/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Building empty aws_iot component due to configuration
-- Component names: soc log heap xtensa-debug-module app_trace freertos vfs newlib esp_ringbuf driver esp_event ethernet mbedtls micro-ecc efuse bootloader_support partition_table app_update spi_flash nvs_flash lwip tcpip_adapter pthread smartconfig_ack wpa_supplicant espcoredump esp32 cxx asio jsmn aws_iot bootloader bt coap console nghttp esp-tls esp_adc_cal tcp_transport esp_http_client esp_http_server esp_https_ota openssl esp_https_server esptool_py expat wear_levelling sdmmc fatfs freemodbus idf_test json libsodium mdns mqtt protobuf-c protocomm spiffs ulp unity wifi_provisioning main
-- Component paths: /Users/Lukas/esp/esp-idf/components/soc;/Users/Lukas/esp/esp-idf/components/log;/Users/Lukas/esp/esp-idf/components/heap;/Users/Lukas/esp/esp-idf/components/xtensa-debug-module;/Users/Lukas/esp/esp-idf/components/app_trace;/Users/Lukas/esp/esp-idf/components/freertos;/Users/Lukas/esp/esp-idf/components/vfs;/Users/Lukas/esp/esp-idf/components/newlib;/Users/Lukas/esp/esp-idf/components/esp_ringbuf;/Users/Lukas/esp/esp-idf/components/driver;/Users/Lukas/esp/esp-idf/components/esp_event;/Users/Lukas/esp/esp-idf/components/ethernet;/Users/Lukas/esp/esp-idf/components/mbedtls;/Users/Lukas/esp/esp-idf/components/micro-ecc;/Users/Lukas/esp/esp-idf/components/efuse;/Users/Lukas/esp/esp-idf/components/bootloader_support;/Users/Lukas/esp/esp-idf/components/partition_table;/Users/Lukas/esp/esp-idf/components/app_update;/Users/Lukas/esp/esp-idf/components/spi_flash;/Users/Lukas/esp/esp-idf/components/nvs_flash;/Users/Lukas/esp/esp-idf/components/lwip;/Users/Lukas/esp/esp-idf/components/tcpip_adapter;/Users/Lukas/esp/esp-idf/components/pthread;/Users/Lukas/esp/esp-idf/components/smartconfig_ack;/Users/Lukas/esp/esp-idf/components/wpa_supplicant;/Users/Lukas/esp/esp-idf/components/espcoredump;/Users/Lukas/esp/esp-idf/components/esp32;/Users/Lukas/esp/esp-idf/components/cxx;/Users/Lukas/esp/esp-idf/components/asio;/Users/Lukas/esp/esp-idf/components/jsmn;/Users/Lukas/esp/esp-idf/components/aws_iot;/Users/Lukas/esp/esp-idf/components/bootloader;/Users/Lukas/esp/esp-idf/components/bt;/Users/Lukas/esp/esp-idf/components/coap;/Users/Lukas/esp/esp-idf/components/console;/Users/Lukas/esp/esp-idf/components/nghttp;/Users/Lukas/esp/esp-idf/components/esp-tls;/Users/Lukas/esp/esp-idf/components/esp_adc_cal;/Users/Lukas/esp/esp-idf/components/tcp_transport;/Users/Lukas/esp/esp-idf/components/esp_http_client;/Users/Lukas/esp/esp-idf/components/esp_http_server;/Users/Lukas/esp/esp-idf/components/esp_https_ota;/Users/Lukas/esp/esp-idf/components/openssl;/Users/Lukas/esp/esp-idf/components/esp_https_server;/Users/Lukas/esp/esp-idf/components/esptool_py;/Users/Lukas/esp/esp-idf/components/expat;/Users/Lukas/esp/esp-idf/components/wear_levelling;/Users/Lukas/esp/esp-idf/components/sdmmc;/Users/Lukas/esp/esp-idf/components/fatfs;/Users/Lukas/esp/esp-idf/components/freemodbus;/Users/Lukas/esp/esp-idf/components/idf_test;/Users/Lukas/esp/esp-idf/components/json;/Users/Lukas/esp/esp-idf/components/libsodium;/Users/Lukas/esp/esp-idf/components/mdns;/Users/Lukas/esp/esp-idf/components/mqtt;/Users/Lukas/esp/esp-idf/components/protobuf-c;/Users/Lukas/esp/esp-idf/components/protocomm;/Users/Lukas/esp/esp-idf/components/spiffs;/Users/Lukas/esp/esp-idf/components/ulp;/Users/Lukas/esp/esp-idf/components/unity;/Users/Lukas/esp/esp-idf/components/wifi_provisioning;/Users/Lukas/esp/hello_world/main
-- IDF_VER: v3.3
-- Project is not inside a git repository,                         will not use 'git describe' to determine PROJECT_VER.
-- Project version: 1
CMake Deprecation Warning at /Users/Lukas/esp/esp-idf/components/mbedtls/mbedtls/CMakeLists.txt:1 (cmake_minimum_required):
  Compatibility with CMake < 2.8.12 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
[... cut ...] && :
/bin/sh: line 1:  3158 Killed: 9               /Users/Lukas/esp/xtensa-esp32-elf/bin/xtensa-esp32-elf-ranlib esp-idf/xtensa-debug-module/libxtensa-debug-module.a
[33/820] Building C object esp-idf/fre...nt_freertos.dir/FreeRTOS-openocd.c.obj
ninja: build stopped: subcommand failed.
ninja failed with exit code 1
```

I am not too sure yet what this error is yet but I will be working on it and update what I found.

I hope this thread is going to help solve a lot of problems for macos users :)

krupis, 2021-01-29:
I assume in order to compile grblHAL project I have to enter my drivers folder:

grblHAL -> drivers -> ESP32 

[image]

As expected, the initial attempt to build the project fails with the error message:
```
Lukass-Air:ESP32 Lukas$ idf.py -p /dev/cu.usbserial-0001 flash
Note: You are using Python 3.9.1. Python 3 support is new, please report any problems you encounter. Search for 'Setting the Python Interpreter' in the ESP-IDF docs if you want to use Python 2.7.
Checking Python dependencies...
Python requirements from /Users/Lukas/esp/esp-idf/requirements.txt are satisfied.
Running ninja in directory /Users/Lukas/Downloads/grblHAL-master/drivers/ESP32/build
Executing "ninja all"...
[0/1] Re-running CMake...
-- Building for target esp32
-- Building empty aws_iot component due to configuration
-- Component names: soc log heap freertos vfs newlib esp_ringbuf driver esp_event ethernet mbedtls micro-ecc efuse bootloader_support partition_table app_update spi_flash nvs_flash lwip tcpip_adapter pthread smartconfig_ack wpa_supplicant xtensa-debug-module espcoredump esp32 cxx app_trace asio jsmn aws_iot bootloader bt coap console nghttp esp-tls esp_adc_cal tcp_transport esp_http_client esp_http_server esp_https_ota openssl esp_https_server esptool_py expat wear_levelling sdmmc fatfs freemodbus idf_test json libsodium mdns mqtt protobuf-c protocomm spiffs ulp unity wifi_provisioning
-- Component paths: /Users/Lukas/esp/esp-idf/components/soc;/Users/Lukas/esp/esp-idf/components/log;/Users/Lukas/esp/esp-idf/components/heap;/Users/Lukas/esp/esp-idf/components/freertos;/Users/Lukas/esp/esp-idf/components/vfs;/Users/Lukas/esp/esp-idf/components/newlib;/Users/Lukas/esp/esp-idf/components/esp_ringbuf;/Users/Lukas/esp/esp-idf/components/driver;/Users/Lukas/esp/esp-idf/components/esp_event;/Users/Lukas/esp/esp-idf/components/ethernet;/Users/Lukas/esp/esp-idf/components/mbedtls;/Users/Lukas/esp/esp-idf/components/micro-ecc;/Users/Lukas/esp/esp-idf/components/efuse;/Users/Lukas/esp/esp-idf/components/bootloader_support;/Users/Lukas/esp/esp-idf/components/partition_table;/Users/Lukas/esp/esp-idf/components/app_update;/Users/Lukas/esp/esp-idf/components/spi_flash;/Users/Lukas/esp/esp-idf/components/nvs_flash;/Users/Lukas/esp/esp-idf/components/lwip;/Users/Lukas/esp/esp-idf/components/tcpip_adapter;/Users/Lukas/esp/esp-idf/components/pthread;/Users/Lukas/esp/esp-idf/components/smartconfig_ack;/Users/Lukas/esp/esp-idf/components/wpa_supplicant;/Users/Lukas/esp/esp-idf/components/xtensa-debug-module;/Users/Lukas/esp/esp-idf/components/espcoredump;/Users/Lukas/esp/esp-idf/components/esp32;/Users/Lukas/esp/esp-idf/components/cxx;/Users/Lukas/esp/esp-idf/components/app_trace;/Users/Lukas/esp/esp-idf/components/asio;/Users/Lukas/esp/esp-idf/components/jsmn;/Users/Lukas/esp/esp-idf/components/aws_iot;/Users/Lukas/esp/esp-idf/components/bootloader;/Users/Lukas/esp/esp-idf/components/bt;/Users/Lukas/esp/esp-idf/components/coap;/Users/Lukas/esp/esp-idf/components/console;/Users/Lukas/esp/esp-idf/components/nghttp;/Users/Lukas/esp/esp-idf/components/esp-tls;/Users/Lukas/esp/esp-idf/components/esp_adc_cal;/Users/Lukas/esp/esp-idf/components/tcp_transport;/Users/Lukas/esp/esp-idf/components/esp_http_client;/Users/Lukas/esp/esp-idf/components/esp_http_server;/Users/Lukas/esp/esp-idf/components/esp_https_ota;/Users/Lukas/esp/esp-idf/components/openssl;/Users/Lukas/esp/esp-idf/components/esp_https_server;/Users/Lukas/esp/esp-idf/components/esptool_py;/Users/Lukas/esp/esp-idf/components/expat;/Users/Lukas/esp/esp-idf/components/wear_levelling;/Users/Lukas/esp/esp-idf/components/sdmmc;/Users/Lukas/esp/esp-idf/components/fatfs;/Users/Lukas/esp/esp-idf/components/freemodbus;/Users/Lukas/esp/esp-idf/components/idf_test;/Users/Lukas/esp/esp-idf/components/json;/Users/Lukas/esp/esp-idf/components/libsodium;/Users/Lukas/esp/esp-idf/components/mdns;/Users/Lukas/esp/esp-idf/components/mqtt;/Users/Lukas/esp/esp-idf/components/protobuf-c;/Users/Lukas/esp/esp-idf/components/protocomm;/Users/Lukas/esp/esp-idf/components/spiffs;/Users/Lukas/esp/esp-idf/components/ulp;/Users/Lukas/esp/esp-idf/components/unity;/Users/Lukas/esp/esp-idf/components/wifi_provisioning
warning: ignored attempt to assign user value to IDF_CMAKE, which gets its value from the environment
CMake Warning at /Users/Lukas/esp/esp-idf/tools/cmake/crosstool_version_check.cmake:8 (message):
  Xtensa toolchain
  /Users/Lukas/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
  version 8.4.0 is not the supported version 5.2.0.  Check Getting Started
  documentation or proceed at own risk.
Call Stack (most recent call first):
  /Users/Lukas/esp/esp-idf/tools/cmake/idf_functions.cmake:213 (gcc_version_check)
  /Users/Lukas/esp/esp-idf/CMakeLists.txt:74 (idf_verify_environment)

CMake Warning at /Users/Lukas/esp/esp-idf/tools/cmake/crosstool_version_check.cmake:25 (message):
  Xtensa toolchain
  /Users/Lukas/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
  does not appear to be built with crosstool-ng.  Check Getting Started
  documentation or proceed at own risk.
Call Stack (most recent call first):
  /Users/Lukas/esp/esp-idf/tools/cmake/idf_functions.cmake:214 (crosstool_version_check)
  /Users/Lukas/esp/esp-idf/CMakeLists.txt:74 (idf_verify_environment)

-- IDF_VER: v3.3
-- Project is not inside a git repository,                         will not use 'git describe' to determine PROJECT_VER.
[... cut ...]
CMake Generate step failed.  Build files cannot be regenerated correctly.
FAILED: build.ninja 
/Users/Lukas/.espressif/tools/cmake/3.16.4/CMake.app/Contents/bin/cmake -S/Users/Lukas/Downloads/grblHAL-master/drivers/ESP32 -B/Users/Lukas/Downloads/grblHAL-master/drivers/ESP32/build
ninja: error: rebuilding 'build.ninja': subcommand failed
ninja failed with exit code 1
Lukass-Air:ESP32 Lukas$ 
```

This time, it seems that it is complaining about ; 
Make Generate step failed.  Build files cannot be regenerated correctly.

terjeio, 2021-01-29:
Have you searched (or asked) the [ESP-IDF forum](https://esp32.com/viewforum.php?f=13) for any clues? There is also a github repository with a rather large [issues section](https://github.com/espressif/esp-idf/issues)...

> I assume in order to compile grblHAL project I have to enter my drivers folder:

Yes. I think you are getting closer now. The error about missing _eeprom/eeprom_24LC16B.c_ is for plugin code that has to be copied to the driver. See the [readme.md](https://github.com/terjeio/grblHAL/blob/master/drivers/ESP32/eeprom/README.md) file in the eeprom folder.

However, the [test branch](https://github.com/terjeio/grblHAL/tree/test) is easier to set up as this has all files in place, no copying is needed. I have also made some changes to [CMakeLists.txt](https://github.com/terjeio/grblHAL/blob/test/drivers/ESP32/CMakeLists.txt) to simplify configuration.

krupis, 2021-01-29:
I have not yet asked the ESP-IDF forum yet though I have always searched there for any clues. ESP-IDF forum will be my next target to ask for help if I get stuck.. You are also right, I completely missed that step regarding copying the plugin code for the EEPROM inside drivers.

After I have done copying the plugin EEPROM files to my drivers/eeprom folder, I am getting a bunch of new errors:

'''
Lukass-Air:~ Lukas$ cd ~/Downloads/grblHAL-master/drivers/ESP32
Lukass-Air:ESP32 Lukas$ idf.py build
Note: You are using Python 3.9.1. Python 3 support is new, please report any problems you encounter. Search for 'Setting the Python Interpreter' in the ESP-IDF docs if you want to use Python 2.7.
Checking Python dependencies...
Python requirements from /Users/Lukas/esp/esp-idf/requirements.txt are satisfied.
Running ninja in directory /Users/Lukas/Downloads/grblHAL-master/drivers/ESP32/build
Executing "ninja all"...
ninja: error: '../eeprom/eeprom_24LC16B.c', needed by 'CMakeFiles/grbl.elf.dir/eeprom/eeprom_24LC16B.c.obj', missing and no known rule to make it
ninja failed with exit code 1
Lukass-Air:ESP32 Lukas$ idf.py build
Note: You are using Python 3.9.1. Python 3 support is new, please report any problems you encounter. Search for 'Setting the Python Interpreter' in the ESP-IDF docs if you want to use Python 2.7.
Checking Python dependencies...
Python requirements from /Users/Lukas/esp/esp-idf/requirements.txt are satisfied.
Running ninja in directory /Users/Lukas/Downloads/grblHAL-master/drivers/ESP32/build
Executing "ninja all"...
[1/1082] Building C object esp-idf/soc/CMakeFiles/idf_component_soc.dir/esp32/cpu_util.c.obj
FAILED: esp-idf/soc/CMakeFiles/idf_component_soc.dir/esp32/cpu_util.c.obj 
/Users/Lukas/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc -DESP_PLATFORM -DGCC_NOT_5_2_0=1 -DHAVE_CONFIG_H -DIDF_VER=\"v3.3\" -I/Users/Lukas/esp/esp-idf/components/soc/esp32/include -I/Users/Lukas/esp/esp-idf/components/soc/include -Iconfig -I/Users/Lukas/esp/esp-idf/components/esp32/include -I/Users/Lukas/esp/esp-idf/components/driver/include -I/Users/Lukas/esp/esp-idf/components/esp_ringbuf/include -I/Users/Lukas/esp/esp-idf/components/tcpip_adapter/include -I/Users/Lukas/esp/esp-idf/components/lwip/include/apps -I/Users/Lukas/esp/esp-idf/components/lwip/lwip/src/include -I/Users/Lukas/esp/esp-idf/components/lwip/port/esp32/include -I/Users/Lukas/esp/esp-idf/components/lwip/port/esp32/include/arch -I/Users/Lukas/esp/esp-idf/components/lwip/include_compat -I/Users/Lukas/esp/esp-idf/components/vfs/include -I/Users/Lukas/esp/esp-idf/components/esp_event/include -I/Users/Lukas/esp/esp-idf/components/log/include -I/Users/Lukas/esp/esp-idf/components/efuse/include -I/Users/Lukas/esp/esp-idf/components/efuse/esp32/include -I/Users/Lukas/esp/esp-idf/components/newlib/platform_include -I/Users/Lukas/esp/esp-idf/components/newlib/include -I/Users/Lukas/esp/esp-idf/components/freertos/include -I/Users/Lukas/esp/esp-idf/components/app_trace/include -I/Users/Lukas/esp/esp-idf/components/heap/include -mlongcalls -Wno-frame-address   "-mlongcalls -Wno-frame-address" -Og -ffunction-sections -fdata-sections -fstrict-volatile-bitfields -nostdlib -Wall -Werror=all -Wno-error=unused-function -Wno-error=unused-but-set-variable -Wno-error=unused-variable -Wno-error=deprecated-declarations -Wextra -Wno-unused-parameter -Wno-sign-compare -ggdb -std=gnu99 -Wno-old-style-declaration -MD -MT esp-idf/soc/CMakeFiles/idf_component_soc.dir/esp32/cpu_util.c.obj -MF esp-idf/soc/CMakeFiles/idf_component_soc.dir/esp32/cpu_util.c.obj.d -o esp-idf/soc/CMakeFiles/idf_component_soc.dir/esp32/cpu_util.c.obj   -c /Users/Lukas/esp/esp-idf/components/soc/esp32/cpu_util.c
xtensa-esp32-elf-gcc: error: unrecognized command line option '-mlongcalls -Wno-frame-address'; did you mean '--warn-no-frame-address'?
[2/1082] Building C object esp-idf/soc/CMakeFiles/idf_component_soc.dir/esp32/gpio_periph.c.obj
FAILED: esp-idf/soc/CMakeFiles/idf_component_soc.dir/esp32/gpio_periph.c.obj 
/Users/Lukas/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc -DESP_PLATFORM -DGCC_NOT_5_2_0=1 -DHAVE_CONFIG_H -DIDF_VER=\"v3.3\" -I/Users/Lukas/esp/esp-idf/components/soc/esp32/include -I/Users/Lukas/esp/esp-idf/components/soc/include -Iconfig -I/Users/Lukas/esp/esp-idf/components/esp32/include -I/Users/Lukas/esp/esp-idf/components/driver/include -I/Users/Lukas/esp/esp-idf/components/esp_ringbuf/include -I/Users/Lukas/esp/esp-idf/components/tcpip_adapter/include -I/Users/Lukas/esp/esp-idf/components/lwip/include/apps -I/Users/Lukas/esp/esp-idf/components/lwip/lwip/src/include -I/Users/Lukas/esp/esp-idf/components/lwip/port/esp32/include -I/Users/Lukas/esp/esp-idf/components/lwip/port/esp32/include/arch -I/Users/Lukas/esp/esp-idf/components/lwip/include_compat -I/Users/Lukas/esp/esp-idf/components/vfs/include -I/Users/Lukas/esp/esp-idf/components/esp_event/include -I/Users/Lukas/esp/esp-idf/components/log/include -I/Users/Lukas/esp/esp-idf/components/efuse/include -I/Users/Lukas/esp/esp-idf/components/efuse/esp32/include -I/Users/Lukas/esp/esp-idf/components/newlib/platform_include -I/Users/Lukas/esp/esp-idf/components/newlib/include -I/Users/Lukas/esp/esp-idf/components/freertos/include -I/Users/Lukas/esp/esp-idf/components/app_trace/include -I/Users/Lukas/esp/esp-idf/components/heap/include -mlongcalls -Wno-frame-address   "-mlongcalls -Wno-frame-address" -Og -ffunction-sections -fdata- [... cut ...] esp-idf/soc/CMakeFiles/idf_component_soc.dir/esp32/rtc_periph.c.obj.d -o esp-idf/soc/CMakeFiles/idf_component_soc.dir/esp32/rtc_periph.c.obj   -c /Users/Lukas/esp/esp-idf/components/soc/esp32/rtc_periph.c
xtensa-esp32-elf-gcc: error: unrecognized command line option '-mlongcalls -Wno-frame-address'; did you mean '--warn-no-frame-address'?
ninja: build stopped: subcommand failed.
ninja failed with exit code 1
Lukass-Air:ESP32 Lukas$ 
'''

Are you fammiliar with this --warn-no-frame-address thing?

krupis, 2021-01-29:
I am able to compile and flash the grblHAL-test version to my esp32 device without any issues!

terjeio, 2021-01-29:
> Are you fammiliar with this --warn-no-frame-address thing?

No, I wonder if the build folder create in driver folder from the attempt to compile with v4 can have left files hat confuses v3. I just deleted my build folder and rebuilt from scratch so that is a way to ensure a clean build. `idf.py clean` may do the same?

> I am able to compile and flash the grblHAL-test version to my esp32 device without any issues!

Great to hear!

krupis, 2021-01-29:
Nope. Same error after cleaning the build files :/

Are there any reasons why I wouldnt want to use test branch?

terjeio, 2021-01-29:
> Are there any reasons why I wouldnt want to use test branch?

Can't think of any, there are several users running machines with it. It is close to be moved to the master branch. Most changes are for new settings and $-command handling plus simplifications/improvements for the HAL API. I still have some challenges to sort out for the settings subsystem, but these will not affect day-to-day running of a machine. 

> I assume test branch is almost identical to the master branch except that the configuration steps as I mentioned above are already done.

The handling of G-code is almost identical, a bit of refactoring and some minor bug fixes. The settings subsystem has gotten a complete overhaul with many new features, you may try (from a terminal window, not a sender):

`$help` - for a list of help options, e.g. `$help spindle` - for a list of spindle settings with some details.
`$$=<n>`- for help for setting `<n>`, eg `$$=10`
`$<n>` - for displaying the current value of a setting, eg `$10`.
 
There are also new $-commands for enumerating available error codes, alarms, settings and setting groups so that senders can dynamically change their UI to match what is available in the controller. Most of this is to better support plugin code that may add to all these.

Many of the changes are for making plugin code for both myself and 3rd-party developers easier.

More details can be found in the [changelog](https://github.com/terjeio/grblHAL/blob/master/changelog.md).

krupis, 2021-01-29:
Thank you very much you been very helpful! I will now attempt to dig a bit deeper into the project since I have managed to build. 

I assume all the GPIO pinout configurations can be found in:
cnc_boostermap_map.h

[image]

and I can modify them as I like depending on where I connect my things.

terjeio, 2021-01-29:
> I assume all the GPIO pinout configurations can be found in:
cnc_boostermap_map.h

Not so, there are several *_map.h files, the one used for compilation is selected in [CMakeLists.txt](https://github.com/terjeio/grblHAL/blob/test/drivers/ESP32/CMakeLists.txt). There are other configuration switches in this file you may want to check out as well.

> and I can modify them as I like depending on where I connect my things.

Perhaps generic_map.h is a better choice? If you enable the CNC BoosterPack map it is assumed you have an EEPROM/FRAM connected for settings storage - so IMO this is a bad choice unless you have that.

krupis, 2021-01-29:
Of course.. Thanks for clearing that out. 

I have Rbpi running bCNC software. I assume you are fammiliar with this setup. Just last thing that I need to clarify. How do these two communicate between each other? I assume just through USB cable the same way I use to program my ESP32 device using my MACos

terjeio, 2021-01-29:
>  I assume just through USB cable the same way I use to program my ESP32 device using my MACos

Yes, that is the default way of connect. With WiFi enabled should be able to connect via a [socket connection](https://github.com/vlachoudis/bCNC/wiki/Connecting#directly-in-bcnc) too.

krupis, 2021-01-29:
I was just going through different options in Cmakelist and noticed webui which sounded quite interesting. I initially assumed that this will allow me to monitor/change the settings over the web interface. I have modified 3 things in cmakefile:

1. OPTION(Networking "Wifi + protocols" ON) set to ON
2. OPTION(WebUI "WebUI services" ON) set to ON
3. OPTION(SoftAP "Enable soft AP mode" ON)

However, I am getting an error:
```
[26/51] Building C object CMakeFiles/grbl.elf.dir/web/backend.c.obj
FAILED: CMakeFiles/grbl.elf.dir/web/backend.c.obj 
/Users/Lukas/esp/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc -DAUTH_ENABLE -DBOARD_BDRING_V4 -DESP_PLATFORM -DGCC_NOT_5_2_0=0 -DGRBL_ESP32 -DHAVE_CONFIG_H -DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\" -DNETWORKING_ENABLE -DOVERRIDE_MY_MACHINE -DUNITY_INCLUDE_CONFIG_H -DWEBUI_ENABLE -DWIFI_SOFTAP -DWITH_POSIX -I../. -I/Users/Lukas/esp/esp-idf/components/soc/esp32/include -I/Users/Lukas/esp/esp-idf/components/soc/include -Iconfig -I/Users/Lukas/esp/esp-idf/components/log/include -I/Users/Lukas/esp/esp-idf/components/heap/include -I/Users/Lukas/esp/esp-idf/components/freertos/include -I/Users/Lukas/esp/esp-idf/components/app_trace/include -I/Users/Lukas/esp/esp-idf/components/vfs/include -I/Users/Lukas/esp/esp-idf/components/newlib/platform_include -I/Users/Lukas/esp/esp-idf/components/newlib/include -I/Users/Lukas/esp/esp-idf/components/esp_ringbuf/include -I/Users/Lukas/esp/esp-idf/components/driver/include -I/Users/Lukas/esp/esp-idf/components/esp_event/include -I/Users/Lukas/esp/esp-idf/components/ethernet/include -I/Users/Lukas/esp/esp-idf/components/mbedtls/port/include -I/Users/Lukas/esp/esp-idf/components/mbedtls/mbedtls/include -I/Users/Lukas/esp/esp-idf/components/lwip/include/apps -I/Users/Lukas/esp/esp-idf/components/lwip/lwip/src/include -I/Users/Lukas/esp/esp-idf/components/lwip/port/esp32/include -I/Users/Lukas/esp/esp-idf/components/lwip/port/esp32/include/arch -I/Users/Lukas/esp/esp-idf/components/lwip/include_compat -I/Users/Lukas/esp/esp-idf/components/esp32/include -I/Users/Lukas/esp/esp-idf/components/tcpip_adapter/include -I/Users/Lukas/esp/esp-idf/components/efuse/include -I/Users/Lukas/esp/esp-idf/components/efuse/esp32/include -I/Users/Lukas/esp/esp-idf/components/micro-ecc/micro-ecc -I/Users/Lukas/esp/esp-idf/components/bootloader_support/include -I/Users/Lukas/esp/esp-idf/components/app_update/include -I/Users/Lukas/esp/esp-idf/components/spi_flash/include -I/Users/Lukas/esp/esp-idf/components/nvs_flash/include -I/Users/Lukas/esp/esp-idf/components/pthread/include -I/Users/Lukas/esp/esp-idf/components/smartconfig_ack/include -I/Users/Lukas/esp/esp-idf/components/wpa_supplicant/include -I/Users/Lukas/esp/esp-idf/components/wpa_supplicant/port/include -I/Users/Lukas/esp/esp-idf/components/xtensa-debug-module/include -I/Users/Lukas/esp/esp-idf/components/espcoredump/include -I/Users/Lukas/esp/esp-idf/components/asio/asio/asio/include -I/Users/Lukas/esp/esp-idf/components/asio/port/include -I/Users/Lukas/esp/esp-idf/components/jsmn/include -I/Users/Lukas/esp/esp-idf/components/bt/include -I/Users/Lukas/esp/esp-idf/components/bt/bluedroid/api/include/api -I/Users/Lukas/esp/esp-idf/components/coap/port/include -I/Users/Lukas/esp/esp-idf/components/coap/port/include/coap -I/Users/Lukas/esp/esp-idf/components/coap/libcoap/include -I/Users/Lukas/esp/esp-idf/components/coap/libcoap/include/coap -I/Users/Lukas/esp/esp-idf/components/console -I/Users/Lukas/esp/esp-idf/components/nghttp/port/include -I/Users/Lukas/esp/esp-idf/components/nghttp/nghttp2/lib/includes -I/Users/Lukas/esp/esp-idf/components/esp-tls -I/Users/Lukas/esp/esp-idf/components/esp_adc_cal/include -I/Users/Lukas/esp/esp-idf/components/tcp_transport/include -I/Users/Lukas/esp/esp-idf/components/esp_http_client/include -I/Users/Lukas/esp/esp-idf/components/esp_http_server/include -I/Users/Lukas/esp/esp-idf/components/esp_https_ota/include -I/Users/Lukas/esp/esp-idf/components/openssl/include -I/Users/Lukas/esp/esp-idf/components/esp_https_server/include -I/Users/Lukas/esp/esp-idf/components/expat/expat/expat/lib -I/Users/Lukas/esp/esp-idf/components/expat/port/include -I/Users/Lukas/esp/esp-idf/components/wear_levelling/include -I/Users/Lukas/esp/esp-idf/components/sdmmc/include -I/Users/Lukas/esp/esp-idf/components/fatfs/src -I/Users/Lukas/esp/esp-idf/components/freemodbus/modbus/include -I/Users/Lukas/esp/esp-idf/components/freemodbus/modbus_controller -I/Users/Lukas/esp/esp-idf/components/idf_test/include -I/Users/Lukas/esp/esp-idf/components/json/cJSON -I/Users/Lukas/esp/esp-idf/components/libsodium/libsodium/src/libsodium/include -I/Users/Lukas/esp/esp-idf/components/libsodium/port_include -I/Users/Lukas/esp/esp-idf/components/mdns/include -I/Users/Lukas/esp/esp-idf/components/mqtt/esp-mqtt/include -I/Users/Lukas/esp/esp-idf/components/protobuf-c/protobuf-c -I/Users/Lukas/esp/esp-idf/components/protocomm/include/common -I/Users/Lukas/esp/esp-idf/components/protocomm/include/security -I/Users/Lukas/esp/esp-idf/components/protocomm/include/transports -I/Users/Lukas/esp/esp-idf/components/spiffs/include -I/Users/Lukas/esp/esp-idf/components/ulp/include -I/Users/Lukas/esp/esp-idf/components/unity/include -I/Users/Lukas/esp/esp-idf/components/unity/unity/src -I/Users/Lukas/esp/esp-idf/components/wifi_provisioning/include -mlongcalls -mlongcalls -Os -ffunction-sections -fdata-sections -fstrict-volatile-bitfields -nostdlib -Wall -Werror=all -Wno-error=unused-function -Wno-error=unused-bu [... cut ...]
I can confirm that the option:
OPTION(WebUI "WebUI services" ON)
breaks my code. When I disable this option while keeping other two ON, I am able to compile without issues.

With the other 2 options enabled, I can see WiFi option when typing $HELP and I set the wifi to Access point mode and I can confirm that It have created an access point "GRBL" and I sucesfully made a connection with my phone. I assume in order to do interesting things through WiFi I must have webui option enabled? :)

[image]

terjeio, 2021-01-29:
> I can confirm that the option: OPTION(WebUI "WebUI services" ON) breaks my code

This one is on me, a settings subsystem API change was not implemented in _backend.c._ The test branch has been updated with a fix.

> I assume in order to do interesting things through WiFi I must have webui option enabled? :)

Not neccesarily, you should be able to connect bCNC via a socket connection with only WiFi enabled.
I have not used the WebUI much, you need to attach a SD card and enable that option too for it to be of any real use?

closed 2021-02-07 by terjeio