terjeio/grblHAL #216: Did anyone sucessfully compiled code for Arduino IDE using ESP32

krupis, 2021-01-28:
Please someone share their experience compiling this project for using Arduino IDE.

Using this as a guide:
https://github.com/terjeio/grblHAL/wiki/Compiling-GrblHAL

I am getting stuck at this :

Load the selected grblHAL_xxx folder into the Arduino IDE as a Library.

First of all, there is no such thing as grblHAL_xxx (where xxx is the driver name).

I am not sure what file should I add as a library?

In my grlbHAL-master, there is a drivers folder. Inside drivers, I have ESP32 folder which I assume I have to add as a library:
[image]

After adding this as a library, I can confrim that it appears in arduino library fodlers, but I am not able to confrim the 2nd step;

You can confirm that the library has been added by checking the File > Examples menu, scroll to the bottom of the list where you should see grblHAL for xxx.

In my examples menu, I have got a ESP32 folder, but this one is built in for the ESP32 and contains a lot of basic examples. But thats not the library that I am looking for.

[image]

Did anyone have simmilar or any issues at all compiling this on Arduino IDE? Appreciating any ideas and tips

krupis, 2021-01-28:
I have now found a simmilar port for grbl for ESP32:
https://github.com/bdring/Grbl_Esp32

Are these different grbl firmwares? I can clearly see that this one has got an Arduino type project .ino in it and should be able to compile on arduino?

terjeio, 2021-01-28:
> Are these different grbl firmwares?

Yes.

> I can clearly see that this one has got an Arduino type project .ino in it and should be able to compile on arduino?

You should.

grblHAL is _not_ a Arduino type project, it uses the standard ESP IDF build system. Arduino uses that too so it should be possible to add a .ino-file for grblHAL too depending on which version of ESP IDF Arduino uses.

ESP IDF v4 has a CMake plugin for Eclipse, I will look into converting grblHAL to v4 when I get my new development machine (to be delivered soon). Possibly also adding a .ino file for the Arduino as well. Unfortunately there is no room on my current machine to do that now - disks are full.

krupis, 2021-01-28:
Thanks for clearing that out so fast. Il be still trying to get it to compile on esp-idf v3.3

closed 2021-02-07 by terjeio