terjeio/grblHAL #223: iMXRT1062 Teensy 4.1 Config.h

jerseyguy1996, 2021-01-30:
The following seems like something that I should be changing given the plethora of memory available.  If so, what is a good number to use for both of these?

```
// The number of linear motions in the planner buffer to be planned at any give time. The vast
// majority of RAM that Grbl uses is based on this buffer size. Only increase if there is extra
// available RAM, like when re-compiling for MCU with ample amounts of RAM. Or decrease if the MCU begins to
// crash due to the lack of available RAM or if the CPU is having trouble keeping up with planning
// new incoming motions as they are executed.
// #define BLOCK_BUFFER_SIZE 16 // Uncomment to override default in planner.h.

// Governs the size of the intermediary step segment buffer between the step execution algorithm
// and the planner blocks. Each segment is set of steps executed at a constant velocity over a
// fixed time defined by ACCELERATION_TICKS_PER_SECOND. They are computed such that the planner
// block velocity profile is traced exactly. The size of this buffer governs how much step
// execution lead time there is for other Grbl processes have to compute and do their thing
// before having to come back and refill this buffer, currently at ~50msec of step moves.
// #define SEGMENT_BUFFER_SIZE 6 // Uncomment to override default in stepper.h.
```

phil-barrett, 2021-01-30:
Check [this issue](https://github.com/terjeio/grblHAL/issues/142). TL;DR BLOCK_BUFFER_SIZE of 1024 works well on the teensy 4.1.

jerseyguy1996, 2021-01-30:
> Check [this issue](https://github.com/terjeio/grblHAL/issues/142). TL;DR BLOCK_BUFFER_SIZE of 1024 works well on the teensy 4.1.

Okay so after reading through that thread it looks like short answer is 1024 but long answer is that there are all sorts of other parameters that will affect my speed.  It's just a cheap K40 so there are mechanical limitations with regards to speed that may come into affect before the software limitations.

jerseyguy1996, 2021-01-30:
I don't know if this should go under another issue but it does fall under config.h so I'll ask here.  In the gnea grbl repository there is a definition for USE_SPINDLE_DIR_AS_ENABLE_PIN.  That doesn't seem to exist in this repository.  Is it implemented in some other way?

phil-barrett, 2021-01-30:
grblHAL has enable and dir.  You can choose to use dir as the enable since most laser CAMs uses M4 (CCW).  You can also invert  the sense of enable and dir in Grbl Settings.

closed 2021-02-07 by terjeio