terjeio/grblHAL #228: ESP32 gpio pins wont update after modifying generic_map.h and Y axis not working

krupis, 2021-02-02:
I am using grblhal-test branch and managed to flash the ESP32 device with the grblHAL firmware. I am now testing driving my stepper drivers and noticed very strange behaviour. 

By default, on the generic_map.h step pins are configured as following:
X-axis step - GPIO 12
Y-axis step - GPIO 14
Z-axis step - GPIO 27.

I was testing all axis and noticed that X and Z are not working.

I have then decided to switch some GPIO arround to see whether my GPIO 14 is broken so I have configured generic_map.h as following:

[image]

I have then reflashed my ESP32 device and notice that nothing has changed.
I can still move X axis when connected to GPIO12 and move Z axis when connected to GPIO 27. My changes to generic_map.h did not apply for some reason. Y axis is still not working.

Then I thought that maybe the device is not even flashing properly so I have added the wifi functionality by modifying Cmakelist and can confirm that WIFI has showed up when typing $HELP so the device is flashing properly.

Also, another issue I have noticed , sometimes when I type $HELP in terminal, it just returns me some random numbers that looks like G-CODE instead of returning me actual menu

terjeio, 2021-02-02:
Attach your CMakeLists.txt and generic_map.h here so I can check.

`$help` should not be issued inside a sender app if the output is in any way parsed. Use an external terminal app such as PuTTY instead. I'll have to check the `$help` command output from the ESP32 driver as most of my testing has been with other drivers and I have not experienced any problems with them.

krupis, 2021-02-02:
[CMakeLists.txt](https://github.com/terjeio/grblHAL/files/5910810/CMakeLists.txt)

I am not able to upload generic_map.h so il just copy paste its contents:
```
/*
  generic_map.h - driver code for ESP32

  Part of grblHAL

  Copyright (c) 2020 Terje Io

  Grbl is free software: you can redistribute it and/or modify
  it under the terms of the GNU General Public License as published by
  the Free Software Foundation, either version 3 of the License, or
  (at your option) any later version.

  Grbl is distributed in the hope that it will be useful,
  but WITHOUT ANY WARRANTY; without even the implied warranty of
  MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
  GNU General Public License for more details.

  You should have received a copy of the GNU General Public License
  along with Grbl.  If not, see <http://www.gnu.org/licenses/>.
*/

#ifdef VFD_SPINDLE
#error "Generic board does not have support for VFD spindle."
#endif

#if SDCARD_ENABLE

// Pin mapping when using SPI mode.
// With this mapping, SD card can be used both in SPI and 1-line SD mode.
// Note that a pull-up on CS line is required in SD mode.
#define PIN_NUM_MISO 19
#define PIN_NUM_MOSI 23
#define PIN_NUM_CLK  18
#define PIN_NUM_CS   5

#endif // SDCARD_ENABLE

// timer definitions
#define STEP_TIMER_GROUP TIMER_GROUP_0
#define STEP_TIMER_INDEX TIMER_0

// Define step pulse output pins.
#define X_STEP_PIN  GPIO_NUM_27
#define Y_STEP_PIN  GPIO_NUM_14
#define Z_STEP_PIN  GPIO_NUM_12

// Define step direction output pins. NOTE: All direction pins must be on the same port.
#define X_DIRECTION_PIN GPIO_NUM_26
#define Y_DIRECTION_PIN GPIO_NUM_25
#define Z_DIRECTION_PIN GPIO_NUM_33

// Define stepper driver enable/disable output pin(s).
#define STEPPERS_DISABLE_PIN    GPIO_NUM_13

// Define homing/hard limit switch input pins and limit interrupt vectors.
#define X_LIMIT_PIN GPIO_NUM_2
#define Y_LIMIT_PIN GPIO_NUM_4
#define Z_LIMIT_PIN GPIO_NUM_15

#ifndef VFD_SPINDLE
// Define spindle enable and spindle direction output pins.
#define SPINDLE_ENABLE_PIN      GPIO_NUM_18
#define SPINDLE_DIRECTION_PIN   GPIO_NUM_5
#define SPINDLEPWMPIN           GPIO_NUM_17
#endif

// Define flood and mist coolant enable output pins.

#define COOLANT_FLOOD_PIN   GPIO_NUM_16
#define COOLANT_MIST_PIN    GPIO_NUM_21

// Define user-control CONTROLs (cycle start, reset, feed hold) input pins.
#define RESET_PIN       GPIO_NUM_34
#define FEED_HOLD_PIN   GPIO_NUM_36
#define CYCLE_START_PIN GPIO_NUM_39
#define SAFETY_DOOR_PIN GPIO_NUM_35

// Define probe switch input pin.
#if PROBE_ENABLE
#define PROBE_PIN   GPIO_NUM_32
#endif

#if KEYPAD_ENABLE
#error No free pins for keypad!
#endif

#if IOEXPAND_ENABLE || KEYPAD_ENABLE || EEPROM_ENABLE || (TRINAMIC_ENABLE == 2130 && TRINAMIC_I2C)
// Define I2C port/pins
#define I2C_PORT  I2C_NUM_1
#define I2C_SDA   GPIO_NUM_21
#define I2C_SCL   GPIO_NUM_22
#define I2C_CLOCK 100000
#endif

#if IOEXPAND_ENABLE
typedef union {
    uint8_t mask;
    struct {
        uint8_t spindle_on       :1,
                spindle_dir      :1,
                mist_on          :1,
                flood_on         :1,
                stepper_enable_z :1,
                stepper_enable_x :1,
                stepper_enable_y :1,
                reserved         :1;
    };
} ioexpand_t;
#endif
```

According to the generic_map.
X step should be on GPIO 27 but the motor moves when connected to GPIO 12 (that was default ). For some reason it looks like the generic_map does not overwrite the GPIO pins, maybe there is something else that is setting the pins? 

It is also very strange regarding Y axis . I maybe thought that this might be faulty ESP32 device I have, but I had 2 ESP32 devices and both of them acting exactly the same.

krupis, 2021-02-02:
UPDATE:
I have decided to do something silly. I have connected my stepper driver pulse pin to different GPIO pins of ESP32 and testing driving Y-axis. The Y-axis is moving when pulse is connected to GPIO26. 
This does not make any sense?

krupis, 2021-02-02:
From Cmakelist.txt I have noticed that enabled board type is:
OPTION(BOARD_BDRING_V4 "Compile for bdring v4 3-axis board" ON)
Therefore I must change brding_v4_map.h ! changing generic_map.h does not change my board gpio. So I can confrim that my ESP32 device works as expected as described in bdring_v4_map.h.

Can you explain when do I use generic_map and when to use bdring_v4_map.h and why it is bdring_v4 and not something else?

terjeio, 2021-02-02:
> From Cmakelist.txt I have noticed that enabled board type is:
OPTION(BOARD_BDRING_V4 "Compile for bdring v4 3-axis board" ON)
Therefore I must change brding_v4_map.h ! changing generic_map.h does not change my board gpio. So I can confrim that my ESP32 device works as expected as described in bdring_v4_map.h.

You can change _generic_map.h_ if you leave all boards off, from _CMakeLists.txt_ just above the board options list:

```
# NOTE: Only one board may be enabled!
# If none is enabled pin mappings from generic_map.h will be used.
```

This way of doing it is common to all drivers having a _generic_map.h_ file.

> Can you explain when do I use generic_map and when to use bdring_v4_map.h and why it is bdring_v4 and not something else?

`generic_map.h` is best to use if changes are needed? This since it is not for a specific board. I usually set the default board map to  a known board as this normally does not require any modification. I have been pondering adding a configuration for _my_board_map.h_ with no corresponding file in the source to ensure it does not get overwritten on an update. If selected it has to be created by the user, possibly by copying one of the existing maps.

phil-barrett, 2021-02-02:
A suggestion, create your own map file.  Copy generic and modify?  I don't know about esp32 but for teensy there are just a couple of places that need to be changed for a new map file. Then no overwrite risk (though your other mods obviously are).

krupis, 2021-02-03:
Understood. Thanks a lot for quick follow up. I am now sorting my feed hold, cycle and reset buttons. I am still considering various options for the emergency STOP button. For that, I am considering 2 choices:

1. Connecting the ESP32 reset pin to GND. When the E-STOP is pressed, it will power cycle the microcontroller.

2. Connecting E-STOP to my 24V supply that is powering the stepper motors. What I am concerned about however, when i press the button, it should disable the power to the drivers, but the ESP32 is still running, so when I enable the power to drivers back again, I assume its just going to continue the job immediately ? I am not sure whether thats what I want from E-STOP.

closed 2021-02-10 by krupis