terjeio/grblHAL #244: Where can I find alarm code?

MARIOBASZ, 2021-02-16:
I want to start using Hal. I will try to familiarize myself, the same as with your Grbl GCode Sender. I have tried from bCNC, with which I am comfortable, but I get error 18.
In which file do I find a list of errors?
Thanks.

phil-barrett, 2021-02-16:
There is a CSV file in the [ioSender distribution](https://github.com/terjeio/Grbl-GCode-Sender/releases/download/v028-beta8/GCode.Sender.Beta8.zip).  

I've added it here but the formatting isn't the best. 18 is reset asserted, btw.

Error Code in v1.1+ | Error Message in v1.0- | Error Description
-- | -- | --
1 | Expected command letter | G-code words consist of a letter and a value. Letter was not found.
2 | Bad number format | Missing the expected G-code word value or numeric value format is not valid.
3 | Invalid statement | Grbl '$' system command was not recognized or supported.
4 | Value < 0 | Negative value received for an expected positive value.
5 | Setting disabled | Homing cycle failure. Homing is not enabled via settings.
6 | Value < 3 usec | Minimum step pulse time must be greater than 3usec.
7 | EEPROM read fail. Using defaults | An EEPROM read failed. Auto-restoring affected EEPROM to default values.
8 | Not idle | Grbl '$' command cannot be used unless Grbl is IDLE. Ensures smooth operation during a job.
9 | G-code lock | G-code commands are locked out during alarm or jog state.
10 | Homing not enabled | Soft limits cannot be enabled without homing also enabled.
11 | Line overflow | Max characters per line exceeded. Received command line was not executed.
12 | Step rate > 30kHz | Grbl '$' setting value cause the step rate to exceed the maximum supported.
13 | Check Door | Safety door detected as opened and door state initiated.
14 | Line length exceeded | Build info or startup line exceeded EEPROM line length limit. Line not stored.
15 | Travel exceeded | Jog target exceeds machine travel. Jog command has been ignored.
16 | Invalid jog command | Jog command has no '=' or contains prohibited g-code.
17 | Setting disabled | Laser mode requires PWM output.
18 | Reset asserted |  
19 | Non positive value |  
20 | Unsupported command | Unsupported or invalid g-code command found in block.
21 | Modal group violation | More than one g-code command from same modal group found in block.
22 | Undefined feed rate | Feed rate has not yet been set or is undefined.
23 | Invalid gcode ID:23 | G-code command in block requires an integer value.
24 | Invalid gcode ID:24 | More than one g-code command that requires axis words found in block.
25 | Invalid gcode ID:25 | Repeated g-code word found in block.
26 | Invalid gcode ID:26 | No axis words found in block for g-code command or current modal state which requires them.
27 | Invalid gcode ID:27 | Line number value is invalid.
28 | Invalid gcode ID:28 | G-code command is missing a required value word.
29 | Invalid gcode ID:29 | G59.x work coordinate systems are not supported.
30 | Invalid gcode ID:30 | G53 only allowed with G0 and G1 motion modes.
31 | Invalid gcode ID:31 | Axis words found in block when no command or current modal state uses them.
32 | Invalid gcode ID:32 | G2 and G3 arcs require at least one in-plane axis word.
33 | Invalid gcode ID:33 | Motion command target is invalid.
34 | Invalid gcode ID:34 | Arc radius value is invalid.
35 | Invalid gcode ID:35 | G2 and G3 arcs require at least one in-plane offset word.
36 | Invalid gcode ID:36 | Unused value words found in block.
37 | Invalid gcode ID:37 | G43.1 dynamic tool length offset is not assigned to configured tool length axis.
38 | Invalid gcode ID:38 | Tool number greater than max supported value or undefined tool selected.
39 | Invalid gcode ID:39 | Value out of range.
40 | Invalid gcode ID:40 | G-code command not allowed when tool change is pending.
41 | Invalid gcode ID:41 | Spindle not running when motion commanded in CSS or spindle sync mode.
42 | Invalid gcode ID:42 | Plane must be ZX for threading.
43 | Invalid gcode ID:43 | Max. feed rate exceeded.
44 | Invalid gcode ID:44 | RPM out of range.
45 | Limit switch engaged | Only homing is allowed when a limit switch is engaged.
46 | Homing required | Home machine to continue.
47 | Invalid gcode ID:47 | ATC: current tool is not set. Set current tool with M61.
48 | Invalid gcode ID:48 | Value word conflict.
50 | E-stop | Emergency stop active.
60 | SD Card | SD Card mount failed.
61 | SD Card | SD Card file open/read failed.
62 | SD Card | SD Card directory listing failed.
63 | SD Card | SD Card directory not found.
64 | SD Card | SD Card file empty.
70 | Bluetooth | Bluetooth initalisation failed.

MARIOBASZ, 2021-02-16:
Thanks Phil. I do not know if bCNC is compatible, I have not been able to make movements (virtual yet) from there. 
Hug.

phil-barrett, 2021-02-16:
Yes, bCNC does work with grblHAL.  Make sure you have a recent version. You also need to clear any inputs (like EStop asserted or door open).  Check out the the [First Run page in the Wiki](https://github.com/terjeio/grblHAL/wiki/First-Run-Grbl-Settings).

MARIOBASZ, 2021-02-17:
I still can not:
in bCNC:
Error: 18 [R]
with GrBL Sender it works
I appreciate if you also inform me in which file I find the list of errors

phil-barrett, 2021-02-17:
What hardware are you running on?

MARIOBASZ, 2021-02-17:
Arduino Due

phil-barrett, 2021-02-17:
Is the reset pulled high or low?  what is $14

phil-barrett, 2021-02-17:
Yes, you probably need to invert your reset signal.  In settings, $14 - check (or uncheck) Reset.  If you are building from the Test Branch, it is in Control Signals.

MARIOBASZ, 2021-02-18:
Thanks Phil, my mistakes, it works already.
Hug.

terjeio, 2021-08-21:
In later versions of grblHAL alarms can be enumerated with the [$EA command](https://github.com/grblHAL/core/wiki/Report-extensions#code-sets-and-enumerations). I am considering adding commands for outputting alarms and error codes in Grbl .csv-format as I have done for settings in the latest version.

closed 2021-08-21 by terjeio