terjeio/grblHAL #245: Building ESP32 driver fails (undefined references)

j00ch, 2021-02-17:
Hi,

I want to try this project for my ESP32 board, but when i try to compile i run into issues.

First of all i am on Arch Linux, the lowest version of esp-idf i can use is 4.0.2. Anything lower gives me problems with pip's, downgrading all the dependencies for version 3 would be very difficult (if possible at all).

When i compile with version 4.2 or 4.3 i get near the end and at the linking part i get:

```
/home/user/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: esp-idf/freertos/libfreertos.a(port_common.c.obj):(.literal.main_task+0x14): undefined reference to `app_main'
/home/user/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: esp-idf/freertos/libfreertos.a(port_common.c.obj): in function `main_task':
/home/user/dev/esp/esp-idf-v4.3-beta1/components/freertos/port/port_common.c:129: undefined reference to `app_main'
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
ninja failed with exit code 1
```

Anybody have advice or pointers?

terjeio, 2021-02-17:
The project should be upgraded to use esp-idf 4.x. How much work that is I don't know. Maybe only change main() to app_main()?

I am going to install 4.x on my new development machine so it is on my todo list, but it would be nice if somebody else could do it. If anybody want to try then please start with the [test branch](https://github.com/terjeio/grblHAL/tree/test).

kdomanski, 2021-02-17:
If you cannot (or don't want to) install the old version of the toolchain on your system, you can build using espressif's [official Docker image](https://hub.docker.com/r/espressif/idf). I opened PR #247 that extends the Readme with instructions on how to do a self-contained build in Docker.

j00ch, 2021-02-18:
@kdomanski 
Thanks for that tip, i am completely new to docker. When i run the first command suggested in my grbl folder cmake fails:

```
CMake Error at /opt/esp/idf/tools/cmake/project.cmake:103 (add_executable):
  Cannot find source file:

    eeprom/eeprom_24LC16B.c

  Tried extensions .c .C .c++ .cc .cpp .cxx .cu .m .M .mm .h .hh .h++ .hm
  .hpp .hxx .in .txx
Call Stack (most recent call first):
  CMakeLists.txt:90 (project)

CMake Error at /opt/esp/idf/tools/cmake/project.cmake:103 (add_executable):
  No SOURCES given to target: grbl.elf
Call Stack (most recent call first):
  CMakeLists.txt:90 (project)

```

What am i doing wrong?

kdomanski, 2021-02-18:
Hmm, existing documentation doesn't mention it, but with either way of building if you have EEPROM support enabled (and by default is is enabled in `my_machine.h`) you also need to copy the EEPROM driver from `plugins/eeprom/` to `drivers/ESP32/eeprom/`.

terjeio, 2021-02-18:
Settings in _my_machine.h_ is (or should be) ignored if building with cmake. The test branch does not need any code to be copied and it might be a worth to try with that instead of the master.

j00ch, 2021-02-19:
@kdomanski That seems to help out indeed, i get near the end now:

```
../driver.c: In function 'set_dir_outputs':
../driver.c:485:17: error: 'A_DIRECTION_PIN' undeclared (first use in this function)
     DIGITAL_OUT(A_DIRECTION_PIN, dir_outbits.a);
                 ^
../driver.c:288:48: note: in definition of macro 'DIGITAL_OUT'
 #define DIGITAL_OUT(pin, state) gpio_set_level(pin, state)
                                                ^
../driver.c:485:17: note: each undeclared identifier is reported only once for each function it appears in
     DIGITAL_OUT(A_DIRECTION_PIN, dir_outbits.a);
                 ^
../driver.c:288:48: note: in definition of macro 'DIGITAL_OUT'
 #define DIGITAL_OUT(pin, state) gpio_set_level(pin, state)
                                                ^
[1077/1098] Building C object CMakeFiles/grbl.elf.dir/grbl/gcode.c.obj
ninja: build stopped: subcommand failed.
ninja failed with exit code 1

```
I've built on a clean setup copied grbl and eeprom, modified my_machine.h and config.h. Anything I've overlooked?

kdomanski, 2021-02-20:
> The test branch does not need any code to be copied and it might be a worth to try with that instead of the master.

@terjeio By the way, what's the rason behind the `test` branch being diverged from `master` for over 3 months now and not being merged? I'm a bit confused as to which branch to use in my workshop and which branch to contribute code to.

terjeio, 2021-02-20:
> By the way, what's the reason behind the test branch being diverged from master for over 3 months now and not being merged?

Available time. I have been ready to merge several times but new issues keep popping up. A serious issue with the iMXRT1062 driver seems to be resolved today so I think I just do it...

Please contribute to the test branch. I'll rename that to the "edge" branch soon.

kdomanski, 2021-02-26:
This is now fixed in `master`.

closed 2021-04-30 by terjeio