terjeio/grblHAL #248: ESP32 Probing help

krupis, 2021-02-18:
Hello. I am running grblHAL test branch and have some trouble understanding how probing works. I have purchased OpenBuilds XYZ probe and wired it to +3.3V, GND and signal. 

Keep in mind, that on the signal I do not have any pullup/pulldown resistors - do I need one?  I have learned that I need pullups for the buttons as the internal pullup resistors does not seem to work for the ESP32.

Also, I have noticed that when $6 = 0, the probe is permamently activated ( I an see Status XYZP ) P indicating that Probe is activated?

However, when I set it to $6 = 1, I think it works as expected. I have thought that $6 = 0 is the default configuration and the probe should work like that?

From what I understand $6 controls ACTIVE HIGH / ACTIVE LOW configuration.

$6 = 0 is default (ACTIVE HIGH) and $6 = 1 is ACTIVE LOW.

When I measure the probe signal pin to GND, I measure 3.3V when the clip is not touching the conductive metal , when it touches, it goes down to 0.

terjeio, 2021-02-18:
>  I have learned that I need pullups for the buttons as the internal pullup resistors does not seem to work for the ESP32.

The pins used for inputs does not have internal pullups available so external pullups are required if not actively driven.

> However, when I set it to $6 = 1, I think it works as expected. I have thought that $6 = 0 is the default configuration and the probe should work like that?

> Does that depend on the contruction of the probe itself?

I depends on the probe, so you have to set $6 to match yours. It seems to me that your probe has an active output, if so no pullup is needed. Anyway I does not hurt to add one as leaving the input floating when the probe is not connected is not wise.

krupis, 2021-02-18:
Thanks for clarifying. Il add a pullup on the probe

closed 2021-04-30 by terjeio