terjeio/grblHAL #269: Alarm:10 after pressing E-Stop, but can't reactivate

einencool, 2021-02-27:
Good Morning @terjeio 
I tried your new Sender from your side, and have installed the GRBLHal Version from last week.

Today I tested something and had to press the E-Stop Button, but now I'm not able to delete the "Alarm:10" state...
And I can't move the machine
[image]

terjeio, 2021-02-27:
Seems it is the sender misbehaving. I have uploaded a new edge version - GCode Sender Edge.8.10.zip, can you try that?

einencool, 2021-02-28:
Yes, this seems to work now.

I have seen, that the problem orrurs much stronger with the old Version, when I have for example my motors not powered, and so the E-Stop is still active, and I then started the software.

The clickbox witht the E-Stop alarm didn't occur.
this is now working with the 8.10 sender.

In the beginning with the 8.9 Version I was so happy, that the message didn't come up, but I didn't mention, that there are other problems after that... :-)

closed 2021-02-28 by einencool