terjeio/grblHAL #271: Backlash compensation on Z axis break G38.2

berghauz, 2021-03-01:
When $162=any_positive_value issued G38.2 works not as expected - it moves down to table and after contact (probe triggered) not stopping, but starting moving several times faster right in the table not responding on probe anymore, only power off could stop it.
If clear $162 to zero - G38.2 working well.

terjeio, 2021-03-01:
Can you try with adding some code after these lines in _grbl/motion_control.c_:

https://github.com/terjeio/grblHAL/blob/dd2aa2ba5a29bfbe663cab494e02d02a0831e64e/drivers/IMXRT1062/grblHAL_Teensy4/src/grbl/motion_control.c#L908-L911

so that it will read:

```
    // Reset the stepper and planner buffers to remove the remainder of the probe motion.
    st_reset();             // Reset step segment buffer.
    plan_reset();           // Reset planner buffer. Zero planner positions. Ensure probing motion is cleared.
    plan_sync_position();   // Sync planner position to current machine position.
#ifdef ENABLE_BACKLASH_COMPENSATION
    mc_sync_backlash_position();
#endif
```

berghauz, 2021-03-03:
Now it's good. Ty!

closed 2021-03-03 by berghauz