terjeio/grblHAL #273: SAM3X8E Arduino Due > 3 Axis

frankschirpke, 2021-03-06:
Greetings
I am playing with the Arduino Due configuring more than 3 axis. Where is the number of axis for the Arduino Due defined? How can I check if the compilation with more than 3 axis was successfull?

Best regards Frank

terjeio, 2021-03-06:
> Where is the number of axis for the Arduino Due defined?

In grbl/config.h:

https://github.com/terjeio/grblHAL/blob/dd2aa2ba5a29bfbe663cab494e02d02a0831e64e/drivers/SAM3X8E/grblHAL_Due/src/grbl/config.h#L35

>  How can I check if the compilation with more than 3 axis was successfull?

By checking the real-time report output - it should list coordinate values for all axes. You may also send G-code to move the additional axes - if not compiled in an error will be issued.

frankschirpke, 2021-03-07:
Thanks Terje

found the problem. However I was working in the wrong directory. There was one library in program files and one in my documents. So every change in the settings did not result in a new compilation...

Best regards

Frank

closed 2021-03-07 by frankschirpke