terjeio/grblHAL #278: Zeroing the Work coordinates

raenji-sk, 2021-03-25:
Hi, I got a weird issue. I can not zero the work coordinates by pressing any of the buttons in bCNC ot IO Gcode sender. Got the Teensy board from Phil up and running today and all seems to work OK, but I can't figure out this one. 
I copied all the settings from Arduino GRBL, are there any specific setting in GRBL HAL which might cause this?

raenji-sk, 2021-03-25:
Oh, never mind, I set the flags on status report to report everything and it now works correctly. Although is there any explanation to this? On arduino I had it set to $10=3 and it was OK.

terjeio, 2021-03-25:
grblHAL uses a lot more bits for the $10 setting, many of the new bits replaces compile time options in the Arduino version.

With $10=3 position is reported in machine coordinates and work coordinate offset is not reported, the sender will then be unable to add any offset to the displayed position. It could be argued that I should have added a new setting for the new bits but it is too late to change that now...

raenji-sk, 2021-03-26:
@terjeio OK, thanks for the explanation. It seems to be working fine, just a bit confusing for a new user of grblHAL.

closed 2021-03-26 by raenji-sk