terjeio/grblHAL #296: Initial Spindle Sync Questions...

NeilMarley, 2021-05-05:
Hi,

I have Spindle Sync working generally but I obviously need to spend some time fine-tuning.

i) It's clear at the moment that I have massive overshoot in sync'd feet rates initially, and probably quite a bit of ringing/oscillation before it settles to something resembling the required thread pitch/feed rate. While overshooting and also sometimes while oscillating - my Z-Axis stepper is missing steps i.e. the commanded rate and/or acceleration are exceeding the max settings (capability) for the axis. 
I read in the pre-discussion about how best to do Spindle Sync that the normal acceleration rate wasn't used - I understand this as having a PID controller output then filtered by another linear ramp would be impossible to tune. But it would appear at the moment that the PID loop output to the driver isn't constrained by the Max speed and acceleration limits set for the Axis either?
Note that my setup has quite slow steppers and modest limits (max feed 1200mm/min, accel 20mm/s2) so this may be much more apparent than in other faster set-ups.

ii) What are the ranges/significance for the P,I,&D settings... are they just gain factors i.e. 1 = unity gain for that dimension? And is the Error parameter the dead-band and what are its units?

iii) Minor: After a G33 command, the previously set feed-rate for e.g.  If set by G1 or F is lost (its normally modal) - is this just normal GCode behaviour? I know its best practice to always have the F parameter in commands but its sometimes easier when experimenting via MDI to set it with F & leave it.

Otherwise, very promising progress - thank you very much for your efforts on this.

Neil

terjeio, 2021-05-05:
i) No constraints implemented yet in the PID loop - how to handle that? Perhaps by tuning the PID loop to avoid ringing.
There is a check for max feed rate at the start of a spindle synced motion, if > 0.9 * Z feed rate an error is returned. Other checks should be added?

ii) Start with P = 1, I = 0 and D = 0. Yes, they are gain factors. Dead band is not yet implemented, the PID loop works on angular position (expected vs actual) so the error signal is in mm? Some tuning info[ here](http://www.linuxcnc.org/docs/html/motion/pid-theory.html).

iii) This is per design. Spindle sync feed rate is calculated from the spindle speed and pitch so cannot be used in a meaningful way to set the modal value. I am not sure now if I decided this on my own or if it is from LinuxCNC (or other) documentation. I am not sure it is a good idea to change this behaviour...

NeilMarley, 2021-05-05:
Thanks for the prompt response.

i) It might be possible for an initial check that the starting P determined acceleration also doesn't exceed the axis programmed limit? I understand that I'm at an early stage and with slow axes .... but if the test runs lose steps it makes the evaluation/tuning painfully slow :-).

ii) Understood, will move forward with the traditional 'simple' PID tuning process....

iii) Understood, it's not a problem once known. But my expectation was that the modal feed rate might be returned to the F value before SS motion was started? If it's currently as per standard/common practice, no problem.

Again, thanks.
Neil