terjeio/grblHAL #299: teensy 4.1 sd card commands

karoria, 2021-05-13:
Hi @terjeio @phil-barrett 
I am new here. Thanks for this wonderful project to opensource community. I have got teensy 4.1 and got it flashed with grblHAL successfully with default pin map (as per phil's ethernet breakout board). I can connect it with cncjs with compatibility mode and can jog it successfully. I have also "dry run" some gcode programs and those seem to work well.
I have formatted an sd card of 32 gb class 10 to fat32 and inserted to sd slot and tried to execute commands $FM, $F etc. but they all failed with error:2 (Bad number format)
Am I missing something while flashing?
Best regards,
Ravi (India)

terjeio, 2021-05-13:
Have you enabled SD card support in [my_machine.h](https://github.com/terjeio/grblHAL/blob/master/drivers/IMXRT1062/grblHAL_Teensy4/src/my_machine.h) by removing the comment from `#define SDCARD_ENABLE`?

`$i+` should list the SD-card plugin if active.

karoria, 2021-05-18:
Thanks @terjeio. That worked! Is there any memory limitations for programs run from sd card? I am asking this because some senders hangs when opening files larger than 10-20MB.

terjeio, 2021-05-18:
1. 2 og 4GB? Max filezize is limited by the FatFS filesystem implemetation.
2. The latest version of the firmware and ioSender has upload implemented via the YModem protocol. Note that this is not possible to start a transfer outside of the sender as the initial start character to be sent from the receiver (controller) is not possible to implement unless an input pin is added for it.

karoria, 2021-05-18:
Thanks for prompt reply. Your answer means a 50MB file should work. If I use ethernet breakout board by phil, can it be possible to host sd as ftp server?

terjeio, 2021-05-18:
> If I use ethernet breakout board by phil, can it be possible to host sd as ftp server?

It should be possible and if so it has to be implemented as part of the networking plugin. There are a number of ftp server implementations on top of lwIP that could be used more or less as-is or as a starting point. [This](https://github.com/gezedo/lwftp) is one I found from a quick search. IIRC the lwIP source has an implementation as well. I have no idea of how much work it will be though...

karoria, 2021-05-18:
Thanks a lot.

terjeio, 2021-05-27:
FTP server functionality has been added to the [networking plugin](https://github.com/grblHAL/Plugin_networking) but sadly the SD card library used for the Teensy is buggy so FTP cannot yet be used with this driver. I have a workaround for the library but I'll wait a bit to see if the owner of the library will fix it before I decide what to do. I can post the patched files here if you want to test it.

karoria, 2021-06-06:
Hi @terjeio. Thanks for your efforts! I would love to give it a try if you have already done your part.

terjeio, 2021-06-06:
The SD card library has been fixed and I have committed an update with the latest changes for ftp (and some other stuff).

Note that the FatFs configuration in the library has to be changed, set

`#define FF_FS_RPATH		2`

 in _ffconf.h_ or you wil get a compiler error.

File transfer when running gcode from the SD card is not allowed.

karoria, 2021-06-07:
Thanks a lot.

closed 2021-06-07 by karoria