terjeio/grblHAL #300: mist on ramps_1.6

arlimbad222, 2021-05-15:
drivers/SAM3X8E/grblHAL_Due/src/ramps_1.6_map.h

I can't find the mist output pin on this file anyone can help me. I want to build CNC with Arduino due so I am using ramps_1.6.

terjeio, 2021-05-15:
Looks like analog pin 10 can be used. Add this to the map file:

```
#define COOLANT_MIST_PORT   PIOB
#define COOLANT_MIST_PIN    19  // Due Analog Pin 9
#define COOLANT_MIST_BIT    (1<<COOLANT_MIST_PIN)
```
There are [numerous other pins](https://github.com/grblHAL/SAM3X8E/blob/master/Arduino%20Due%20shield%20pin%20mappings/ramps_1.6_map.md) that could be used as well, even heater outputs for driving a solenoid or relay directly. AFAIK there is no established convention for which pin to use for mist - it is a 3D printer board after all. Please correct me if I am wrong.

arlimbad222, 2021-05-16:
hi @terjeio 

actually, currently, I am using mega2560 so I make one custom PCB design for the mega 2560, and most of the pins are configured as ramps_1.4 so is it possible to change the pin for different inputs and outputs. I am not a programming guy so I have no experience with programming can you please give an example of how to change pins.

thank you for your response have a good day my friend.

terjeio, 2021-05-16:
No programming needed for changing pins (some like the spindle PWM output is limited to a few). Edit the [relevant map file](https://github.com/terjeio/grblHAL/blob/master/drivers/SAM3X8E/grblHAL_Due/src/ramps_1.6_map.h) and change port and pin assignments before recompiling:

```
#define COOLANT_MIST_PORT   PIOB
#define COOLANT_MIST_PIN    19  // Due Analog Pin 9
#define COOLANT_MIST_BIT    (1<<COOLANT_MIST_PIN)
```

If you look at the [board layout](https://github.com/grblHAL/SAM3X8E/blob/master/Arduino%20Due%20shield%20pin%20mappings/ramps_1.6_map.md) you can see shorthand processor pin names, for the one above B.19, this means PIO`B` is the port to assign and 19 is the pin in that port. If you want to use Arduiono digital pin 09 then this is processor pin C.21 (taken from `[09] C.21` in the board layout), the #defines for that then becomes:

```
#define COOLANT_MIST_PORT   PIOC
#define COOLANT_MIST_PIN    21
#define COOLANT_MIST_BIT    (1<<COOLANT_MIST_PIN)
```

arlimbad222, 2021-05-17:
Hi @terjeio 

Thanks for your response. i order the board when it arrives I start testing as soon as possible so give me some time I will taste that code and get back here.

arlimbad222, 2021-05-19:
hi @terjeio 

the board arrives tomorrow by morning and I have a keypad-related question where can I find the keypad pin mapping like I want to jog my machine via a physical button is that possible.

terjeio, 2021-05-19:
The keypad interface is [I2C based](https://github.com/terjeio/I2C-interface-for-4x4-keyboard) and there are definitions for the I2C port in driver.h:

https://github.com/terjeio/grblHAL/blob/9f405bae810fa42c60ebaf2c21c2ff553e46a049/drivers/SAM3X8E/grblHAL_Due/src/driver.h#L133-L149

A third pin is required for a strobe (interrupt) signal and I have not added that to the Ramps map as I have not checked where the I2C pins are routed. 

One or two users has attempted to write I2C keypad code in addtion to my take. Search issues and/or discussions to find those.

Another option would be to write a plugin that takes advantage of the many free pins on the Due. The [current keypad plugin](https://github.com/grblHAL/Plugin_I2C_keypad/blob/master/keypad.c) could be used as a template.

Finally, I have started to rewrite my [MPG & DRO controller](https://github.com/terjeio/GRBL_MPG_DRO_BoosterPack) code for the Pi Pico (RP2040), this too could become an option when completed.

arlimbad222, 2021-05-20:
hi @terjeio 

okay, thanks for the details I am not a coding guy so I'll wait for MPG & DRO controller and again thanks for the response have a nice day....

MARIOBASZ, 2021-06-04:
hello @ arlimbad222
Are you getting a shield for Due?
can you share a file?
Greetings.

arlimbad222, 2021-06-04:
Hi @MARIOBASZ 

I made one prototype pcb for due and grbl hal but currently i am working on it and my project is in testing stage so whenever it was finished, i will share sketch and PCB designs for due.

Have a good day

closed 2022-01-21 by terjeio