terjeio/grblHAL #301: compact 5?

hitchhiker85, 2021-05-19:
hi!

from where you bought your ballscrew?

have you a link?! which size you bought? have you some pictures? i am on the same project!

herte@hotmail.de
please please mail me ;)

thanks!

terjeio, 2021-05-19:
> from where you bought your ballscrew?

From ebay, second hand from Korea - seller [dy-global](https://www.ebay.com/usr/dy-global).

> have you a link?

No, the one I bought for Z is not available anymore it seems. It was listed as _NSK Used W1003+72P-C2Z+430L Precision Mini Ball Screw 1004+430L BSC-I-585=1C61-1_.

I bought the X screw from him as well, and made a new cross slide for it from aluminium. This listing is gone too, it was listed as _ISSOKU Japan Ball screw GTR0802C5+137L AK6+AF6 W/ nut bracket BSC-I-410=B802_.

Good quality even if used, a lot better than the cheap chinese knock-offs.

[image]

hitchhiker85, 2021-05-20:
thank you! good morning :)

can you show me please pictures of the crossslide and how the nut is mounted?

how long the ballscrews?! thank you!!

[IMG_8779]<https://user-images.githubusercontent.com/20260062/118879656-46d62300-b8f1-11eb-8396-6bf8045bc9c1.jpg>

terjeio, 2021-05-20:
[image]

I do not have any pictures with the ballscrew visible, the nut is fastened with two screws under the toolpost.

> how long the ballscrews?

You can get that from the listing details, e.g. _W1003+72P-C2Z+430L_ -> 1003 = 10mm diameter, 3mm pitch; 430L = 430mm long and C2Z is the grade.

hitchhiker85, 2021-05-20:
oh thNk you

how long is the travel on x and z?! thanks

[IMG_8778]<https://user-images.githubusercontent.com/20260062/118925979-e6ba9d80-b93f-11eb-99e7-85a1a8c639a6.jpg>

terjeio, 2021-05-22:
Then I finally have some measurements from the workshop:

X - approx. 60mm travel, nut bracket is 22 mm, thread is 85mm (from my drawings). 
Z - approx. 320mm travel, nut assembly is 40mm, thread is 360mm.

I made new bearing blocks for the Z-axis as the original ones were too high. This to get the stepper (NEMA-23) mounted inside the the lathe. Clearings are tight...

I have files in Vectric VCarve 10 format for the parts if of interest.

hitchhiker85, 2021-05-23:
good morning!

yes i am very interested in cad files.

do you now which thread is used on the crossslide original?

the leadscrew for Z is tr12x1,5.
i order some taps for tr1.5 and will build a new nut which i can adjust.

on the crossslide i will built a new crossslide with tr8x1.5. i have a tap for it.

if this all done i went to ballscrews. but today i start with stepper mounting plates and spindle index with a hall ic and a disc with 12 magnets.

this works peefect on my unimat 3 with mach3.

i think i would make the slide from alloy and brass. 

if you can send me cad files it would be great.
herte@hotmail.de

thanks

terjeio, 2021-05-23:
> do you now which thread is used on the crossslide original?

Standard M6 thread if I remember correctly.

> i think i would make the slide from alloy and brass.

There is a V groove in the cross slide base plate that I somehow managed to mill in my CNC router with a 90 deg. wood bit...

CAD/CAM files in Vectric VCarve v10 format:

[Emco Compact5 cross slide.crv.zip](https://github.com/terjeio/grblHAL/files/6529084/Emco.Compact5.cross.slide.crv.zip)

closed 2022-01-27 by terjeio