terjeio/grblHAL #308: Can grblHAL  A axis be set by angle (step/°)?

viewsat, 2021-06-05:
Hello there:
I use grblHAL A axis, can it be set by angle (step/°)?
Because I use Deskproto to convert gcode A axis to use angle.

gcode :
G93
G1 Z33.000 A36.522 F15.7
G1 Z30.000 A0.000 F15.7
G1 A360.000 F1.6
G94
G1 X75.556 F360
G93
G1 A0.000 F5.3
G94

terjeio, 2021-06-05:
Yes, as long as you do not change units (mm <> inches). Just configure the number of steps per angle unit (degrees, radians...) for the axis. 

There is a pending change (commented out) for a new setting, `$376`, that disables scaling from inches to mm for given axes. I have not pushed this yet as it triggers a settings reset. If you change units you can activate it by enabling the commented code, it is only 5 lines. Search for `rotational` with CTRL-H if using an Eclipse based editor (set _Scope_ to _Enclosing project_ and do not check _Case sensitive_) to find them. I do not know if the Arduino editor allows such searches as I do not use it.

phomann, 2021-06-05:
Would it be better to flag an axis as "linear" or "rotatonal"? That way a rotational axis could also have it's own set of attributes assigned to it such as  step/°, etc.
Cheers,

Peter

terjeio, 2021-06-05:
> Would it be better to flag an axis as "linear" or "rotatonal"?

This is what the new setting does.

`//    { Settings_Axis_Rotational, Group_Stepper, "Rotational axes", NULL, Format_AxisMask ...`

Attributes such as the unit label _can_ be changed but that would have to be done by the sender? Or in cooperation with the controller if the sender uses grblHAL provided settings enumerations. If a user changes the new settings and the unit label is to change immediately then the sender has to do it. Having entirely new settings for rotational axes parameters is going to complicate the controller code for, IMO, no good reason.

terjeio, 2021-08-21:
> Would it be better to flag an axis as "linear" or "rotatonal"? 

The setting for flagging axes (other than X, Y and Z) as rotational been [added](https://github.com/grblHAL/core/blob/master/changelog.md) to build 20210819 (in the new repo).

closed 2021-08-21 by terjeio