terjeio/grblHAL #314: F446RET6 There was a message when I was powered on, but after that I sent $$ and I didn't get any reply?

hankecnc, 2021-06-07:
I have a F446RET6 development board, I downloaded your source code, nothing has been changed, click RElease 446 Nucleo64 to compile, and select GRBL Driver STM32F401.hex in the Release F446 Nucleo64 folder to burn. Received the news,
GrblHAL 1.1f ['$' or'$HELP' for help]
[MSG:'$H'|'$X' to unlock]
, But I send $$ and I can't get any response,
what should I do?

[image]
[image]
[image]

terjeio, 2021-06-07:
It works for me (I dropped the .bin file on the NODE_F446 drive in Explorer to program the board). Does it respond to CTRL-C or when asking for a real-time response with `?`

BTW, here is the .bin I used:

[GRBL Driver STM32F401.zip](https://github.com/terjeio/grblHAL/files/6609162/GRBL.Driver.STM32F401.zip)

terjeio, 2021-06-07:
Ouch, I am sorry - it failed on the second attempt. The dev board uses USART2 for communication, pins PA2 and PA3 which is also used for step outputs in the generic map.  You have to select one of the Nucleo compatible boards by uncommenting it for the compilation:

https://github.com/terjeio/grblHAL/blob/9f405bae810fa42c60ebaf2c21c2ff553e46a049/drivers/STM32F4xx/Inc/my_machine.h#L25-L29

I have to add a test and stop compilation if an incompatible board map is selected...

hankecnc, 2021-06-07:
Thank you!! Thank you very much!!, I followed your method and successfully received a reply, tomorrow I will test grblhal,
[image]

closed 2021-06-07 by hankecnc