terjeio/grblHAL #320: HOLD Command - how to return to Idle when not running gcode.

humneraudio, 2021-08-21:
I'm using bCNC

If I run a gcode file and press hold, it works, and then I can press cycle/start and it carries on - as expected...

However, if I'm not running any code and I accidentally press hold, how do I return back to idle? If I press hold again, nothing happens and it stays in hold, If I press cycle/start, it will clear hold but then immediately begin the gcode file I have loaded into bCNC which isn't ideal either.

Is there a commend I can send to clear hold? I have shortcuts available in bCNC and can map a button if needed.

karoria, 2021-08-22:
Send "~" to return from hold

terjeio, 2021-08-22:
>  If I press cycle/start, it will clear hold but then immediately begin the gcode file I have loaded into bCNC which isn't ideal either.

This is a sender issue, [ioSender](https://github.com/terjeio/ioSender) (which I have written) returns to idle if the loaded program was not running when exiting a hold.

> Is there a commend I can send to clear hold? I have shortcuts available in bCNC and can map a button if needed.

My guess that this is not possible (short of issuing a reset), it is a state transition that has to be coded differently and that has to be changed in the bCNC source?

humneraudio, 2021-08-25:
No worries, thanks for clarifying, I will open this issue to the bCNC channel.

closed 2022-01-21 by terjeio